AnaviTechnology/anavi-docs #2: Failed to find module 'lirc_rpi gpio_in_pin=18 gpio_out_pin=17'

J7mbo, 2017-10-23:
I'm having the following problem:

> Oct 23 17:55:39 raspberrypi systemd-modules-load[773]: Failed to find module 'lirc_rpi gpio_in_pin=18 gpio_out_pin=17'
> Oct 23 17:55:39 raspberrypi systemd[1]: systemd-modules-load.service: Main process exited, code=exited, status=1/FAILURE
> Oct 23 17:55:39 raspberrypi systemd[1]: Failed to start Load Kernel Modules.

This is a brand new, fresh install of Raspian with no additional software apart from those in your [docs](https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-infrared-phat/anavi-infrared-phat.md#setting-up-lirc). Any ideas why I can't progress past this point?

leon-anavi, 2017-10-24:
Hi @J7mbo,

Following the conversation in Twitter, I posting guidelines also here.
 Please make sure you are following [the exact steps for the user's manual](https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-infrared-phat/anavi-infrared-phat.md#setting-up-lirc). The instructions are for Raspbian Stretch, aka the latest version as of the moment. Following the instructions you need to edit /etc/modules, /etc/lirc/hardware.conf, /etc/lirc/lirc_options.conf, /boot/config.txt and finally to reboot the board. 

Best regards,
Leon

J7mbo, 2017-10-25:
Okay, this now works. I was following the user manual. I found that it intermittently recorded the button presses. I will completely do this again from scratch once I've gotten it working to retry the steps and see where it failed.

Turns out there's a *loose connection* somewhere, I've had to use a hairband to push the device forward slightly to make sure the connection the Pi works! I even bought some GPIO hammer headers to make sure it wasn't a header problem - I think the device may be a little faulty. Regardless, the hairband works and I recorded my buttons.

All I need to do now is replay these using `irsend`. Here's what I'm running (I'm trying loads in a row in case that works):

`sudo irsend SEND_ONCE htw KEY_POWER KEY_POWER KEY_POWER KEY_POWER KEY_POWER2`

Unfortunately, nothing happens. I don't see any LEDs light up (should I?), and nothing seems to being transmitted. Any idea how I can see what's going on in this part now?

J7mbo, 2017-10-25:
Also, here's what `systemctl status lircd` shows when I try `irsend`:

> Oct 25 17:49:13 raspberrypi lircd[381]: lircd-0.9.4c[381]: Notice: accepted new client on /var/run/lirc/lircd
> Oct 25 17:49:13 raspberrypi lircd-0.9.4c[381]: Notice: accepted new client on /var/run/lirc/lircd
> Oct 25 17:49:14 raspberrypi lircd[381]: lircd-0.9.4c[381]: Info: removed client
> Oct 25 17:49:14 raspberrypi lircd-0.9.4c[381]: Info: removed client

So it looks like something's triggering! Just no IR I don't think

leon-anavi, 2017-10-26:
Hi @J7mbo,

Each ANAVI Infrared pHAT has been tested carefully twice: first as electric tests of the printed circuit board and after manually on a Raspberry Pi with Raspbian. It is unlikely to have a hardware issue but if you think this is the case you can ship it back for a repair (or a replacement).

What is the name of your LIRC remote control? Is it htw? Can you please paste the output of the command that lists all saved buttons in it? For example, if it is htw:

```
irsend LIST htw ""
```
Sorry for the inconvenience and thank you for supporting the project.

Best regards,
Leon

leon-anavi, 2017-11-12:
Hi @J7mbo, have you resolved the issue that you were facing? Best regards, Leon

J7mbo, 2017-11-17:
Hey Leon! Sorry for the delay. I haven't set this up since, but it does indeed list some commands such as ON and OFF with some codes next to each one. So that worked. If you still need the actual list, I'll set it up this weekend again and get it

leon-anavi, 2017-11-18:
According to my experience sometimes LIRC might generate codes which are not exact during the scanning of the remote control. I am interested to see you are facing a hardware issue or just mismatch in some of the LIRC configurations.

Best regards,
Leon

leon-anavi, 2019-07-08:
I am closing this issue. In the latest versions of Raspbian lirc_rpi has been replaced by with gpio-ir and gpio-ir-tx. Details are available in the updated version of the user's manual: https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-infrared-phat/anavi-infrared-phat.md#setting-up-lirc

Thanks,
Leon

closed 2019-07-08 by leon-anavi