AnaviTechnology/anavi-docs #3: MQTT Username and Password

SteveEdson, 2018-08-14:
I've just received my Anavi light controller and it is awesome, thank you!

Just a quick question, how can I specify a username and password for a custom mqtt connection (from the configuration page at 192.168.4.1).

Thanks!

leon-anavi, 2018-08-14:
Hi @SteveEdson,

Wow, your parcel has been delivered in just 3 working days. I think this is record for the Bulgarian posts :) Thank you for your contribution and support of this project.

Sorry, at the moment this features is not available at the source code of https://github.com/AnaviTechnology/anavi-light-controller-sw. I am adding it to my TODO list. I will keep you updated.

Best regards, Leon

leon-anavi, 2018-08-17:
Hi @SteveEdson,

I have just added support for MQTT username and password in the latest version of https://github.com/AnaviTechnology/anavi-light-controller-sw. It is still an experimental version and I will test it in more details during the weekend.

You can clone the master of anavi-light-controller-sw, compile it in Arduino IDE and flash it on your ANAVI Light Controller. The steps for flashing new firmware on ANAVI Light Controller are explained on several places:
* [YouTube video](https://www.youtube.com/watch?v=DIa7_CQ2T7c)
* [My personal blog](https://www.anavi.org/article/220/)
* [ESP8266 RGB LED Strip Control with ANAVI Light Controller, Arduino, MQTT, and HTML5 by cnx-software](https://www.cnx-software.com/2018/06/17/esp8266-rgb-led-strip-arduino-mqtt-html5/)

Best regards,
Leon

CODeRUS, 2018-08-27:
works for me, thanks!

leon-anavi, 2018-08-28:
Hi @CODeRUS,

Thank you for the feedback. @SteveEdson I hope that you also updated the firmware of ANAVI Light Controller and you now also have support for MQTT username and password. I am closing the issue.

Thanks, Leon

closed 2018-08-28 by leon-anavi

SteveEdson, 2018-08-29:
No problem @leon-anavi, I haven't got round to flashing my ANAVI yet, but I will do at the weekend. Thanks for the support and for a great product!

CODeRUS, 2018-08-30:
There is a bug not loading mqtt username and password. Fixed in PR: https://github.com/AnaviTechnology/anavi-light-controller-sw/pull/2