AnaviTechnology/anavi-docs #4: Home Assistant integration

CODeRUS, 2018-08-27:
Anyone successfully inegrated Anavi Light Controller with Home Assistant? Show me your magic please!

leon-anavi, 2018-08-28:
Hi @CODeRUS,

There are different way and devices on which you can install Home Assistant. Below are the steps with Raspberry Pi and Hassbian, you can you the same approach and adapt it for other devices supported by Home Assistant:

* Install Hassbian on Raspberry Pi

* Run the command below and enable I2C from *Interfacing Options*:

```bash
sudo raspi-config
```

* Install the MQTT broken Mosquitto usign the following [work around for Hassbian 1.31](https://github.com/home-assistant/hassbian-scripts/issues/76):

```bash
sudo curl -o /opt/hassbian/suites/install_mosquitto.sh https://raw.githubusercontent.com/ludeeus/Tools/master/hassbian-config/hotfix/install_mosquitto.sh && sudo hassbian-config install mosquitto

```

* Enable anonymous connection to the MQTT broker by editting the following setting in /etc/mosquitto/mosquitto.conf:

```
allow_anonymous true
```

* Start Mosquitto and enable its automatic launch at boot:

```bash
sudo systemctl enable mosquitto.service
sudo systemctl start mosquitto.service
```

* Replace DEVICEID with the hash copied during the configuration of ANAVI Light Controller and add the following lines to configuration.yaml

```yaml
# MQTT Broker (aka Mosquitto)
mqtt:
  discovery: true
  broker: hassbian.local
  port: 1883

# ANAVI Light Controller
light:
  - platform: mqtt_json
    name: "ANAVI Light Controller"
    command_topic: "DEVICEID/action/rgbled"
    brightness: true
    rgb: true
```

* Run the following command to restart Home Assistant and reload configurations:

```bash
sudo systemctl restart home-assistant@homeassistant.service
```

Best regards, Leon

CODeRUS, 2018-08-28:
okay, this kinda works, but i'd suggest to change topics to cmnd/deviceid/power to follow common iot mqtt practices. also it would be cool if device send retained stats to stat/deviceid/power. i think i can make a pr.

leon-anavi, 2018-08-28:
Cool, thank you for the feedback. It seems that you got it working so I am closing this particular issue. 

The source code is [here](https://github.com/AnaviTechnology/anavi-light-controller-sw) and GitHub pull requests are more than welcome :) 

Thanks, Leon

closed 2018-08-28 by leon-anavi