AnaviTechnology/anavi-docs #15: Initial flash load fails on OS X : warning: espcomm_sync failed

nscooling, 2019-05-09:
Cannot get past this issue.
Error message:
Arduino: 1.8.9 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, Flash, Disabled, ck, 26 MHz, 40MHz, DIO, 4M (1M SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Sketch uses 263416 bytes (25%) of program storage space. Maximum is 1044464 bytes.
Global variables use 26848 bytes (32%) of dynamic memory, leaving 55072 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

nscooling, 2019-05-09:
Solved. The Rx & Tx were swapped (duh!)

closed 2019-05-09 by nscooling

leon-anavi, 2019-05-09:
Hi @nscooling, glad you solved it so quick.

Just for the record, if anyone else is having issues in future, here is a link to the YouTube video tutorial: https://www.youtube.com/watch?v=HMIkPuz0ZJs

Thanks,
Leon