AnaviTechnology/anavi-docs #20: Getting "hardware does not support sending" and "Cannot initiate device /dev/lirc0"

AndrewD93, 2019-06-06:
I've been using Anavi pHat for one year, I've made a remote controller for my air conditioner and it worked fine. Some days ago I realised it's no longer transmitting IR to the AC and I noticed I'm getting "hardware does not support sending" when running irsend. I've taken a look to see if any update had broken the configuration, but only noticed that hardware.conf was removed, so I created it back and rebooted, but still getting the same error. (Other files as described in user manual still exist and have the right lines)
If I try to run `# systemctl stop lircd` and mode2 and `mode2 -d /dev/lirc0` I'm getting "Cannot initiate device /dev/lirc0".
Meanwhile, I notice that temperature and humidity sensor, which is attached to the hat, keeps working fine as I'm getting correct data on my homemade app.
What can I do?

leon-anavi, 2019-06-06:
Hi @AndrewD93,

Have you recently updated the Linux distribution that you are using? If you are with newer kernel, for example 4.19, lirc-rpi has been replaced by gpio-ir. In /boot/config.txt it should be something like: 

```
dtoverlay=gpio-ir-tx,gpio_pin=17 
dtoverlay=gpio-ir,gpio_pin=18 
```

As soon as I have some spare time I will update the user's manual but it is probably going to take a week more.

Thanks,
Leon

AndrewD93, 2019-06-06:
Thank you, Leon. I've updated the file and rebooted and I'm not getting device errors anymore. However, the command is still not being received by the AC. The remote configuration file is still there, in fact I'm not getting any error from CL like the signal was correctly sent. Maybe changing the module has broken that file or manipulated the signal? Consider that commands are mapped as row signals because they didn't work otherwise.

AndrewD93, 2019-06-07:
Is it possible to manually add lirc-rpi so that I don't need to rollback the distro?

AndrewD93, 2019-06-07:
I've eventually rolled back to 4.14 and now it works. I'll wait for updates. Thank you :)

leon-anavi, 2019-07-08:
Hi @AndrewD93,

The setup for LIRC and ANAVI Infrared pHAT on Raspbian Buster has been completely changed. The exact steps are now available at the updated user's manual: https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-infrared-phat/anavi-infrared-phat.md#setting-up-lirc

Thanks,
Leon

closed 2019-07-08 by leon-anavi

AndrewD93, 2019-07-23:
Do I have to map again my whole remote? After I updated lirc following the new guide and rebooted, I'm no longer getting hardware error but commands are not receipted by target device.

leon-anavi, 2019-07-24:
Hm, sorry, I am not sure. Could you please double check your condigurations first? If it still doesn't work as expected, could you try rescanning the remote control?

As you see there is significant change between Raspbian Stretch and Buster not only because of the kernel version but also because of the LIRC version. But in general the infrared codes should remain the same.

Best regards, Leon

AndrewD93, 2019-07-24:
So now I'm trying to scan the remote again and I noticed that /dev/lirc1 does not exist. I've tried resetting config files as explained in the procedure but that's it.

leon-anavi, 2019-07-24:
Please note that in Raspbian Buster with LIRC 0.10 device /dev/lirc**1** is the receiver and device /dev/lirc**0** is the transmitter. Initially /dev/lirc1 in /etc/lirc/lirc_options.conf is used to scan a remote control. After that the configuration  has to be updated to /dev/lirc0 in order to send infrared commands. LIRC has to be restarted after a change in /etc/lirc/lirc_options.conf/. Also please [make sure  /boot/config.txt is with properly configured gpio-ir-tx as explained in user's manual](https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-infrared-phat/anavi-infrared-phat.md#setting-up-lirc).

Best regards, Leon

AndrewD93, 2019-07-24:
I did update the lirc configuration file and rebooted. This is what I get:
```
pi@RasPi3:~ $ sudo systemctl stop lircd
Warning: Stopping lircd.service, but it can still be activated by: lircd.socket
pi@RasPi3:~ $ mode2 -d /dev/lirc1
Using driver default on device /dev/lirc1
Cannot initiate device /dev/lirc1
```

leon-anavi, 2019-07-24:
Hm, probably something is missing from the configurations. Could you please double check your configurations in /boot/config.txt and /etc/lirc/lirc_options.conf ?

Thanks, Leon

AndrewD93, 2019-07-24:
```
dtoverlay=gpio-ir,gpio_in_pin=18
dtoverlay=gpio-ir-tx,gpio_out_pin=17
```
is at the end of /boot/config.txt and here is lirc_options:
```
[lircd]
nodaemon        = False
driver          = default
device          = /dev/lirc1
output          = /var/run/lirc/lircd
pidfile         = /var/run/lirc/lircd.pid
plugindir       = /usr/lib/arm-linux-gnueabihf/lirc/plugins                                                             permission      = 666
allow-simulate  = No
repeat-max      = 600
[lircmd]
uinput          = False
nodaemon        = False
```
If I try to ls /dev there's lirc0 but no lirc1

leon-anavi, 2019-07-24:
In my opinion there is a typo in your  /boot/config.txt: you have gpio_in_pin instead of gpio_pin. This **in** is not needed and breaks the configurations. Same for **out** for the next line with gpio-ir-tx. Could you please double check and fix it?

Thanks,
Leon

AndrewD93, 2019-07-24:
Alright that was it. Now everything works as expected. Sorry for bothering you and thanks for all the support.

Best regards
Andrea

leon-anavi, 2019-07-24:
No worries, I am glad that I was able to help and you have your system up and running :) LIRC is a very powerful tool but unfortunately the whole configuration is difficult.

Best regards, Leon

SanerBCN, 2020-03-31:
> Hi @AndrewD93,
> 
> Have you recently updated the Linux distribution that you are using? If you are with newer kernel, for example 4.19, lirc-rpi has been replaced by gpio-ir. In /boot/config.txt it should be something like:
> 
> ```
> dtoverlay=gpio-ir-tx,gpio_pin=17 
> dtoverlay=gpio-ir,gpio_pin=18 
> ```

After a lot of issues in trying to get LIRC installed on a Buster version of Rasbian, I finally figured out that the order in which you put thoses lines in the config.txt is VERY important. I had the tx-pin second and that didn't work until i moved it tot he first place!!!

Hope this saves somebody time :-)

leon-anavi, 2020-03-31:
Hi @SanerBCN,

Thank you for the feedback. Could you please share your exact `dtoverlay` settings in `/boot/config.txt` so that I can update the user's manual and notes?

Thanks,
Leon

SanerBCN, 2020-03-31:
`# Uncomment this to enable the lirc-rpi module

# Working config for the IR
# tx pin needs to be declared FIRST!!!
dtoverlay=gpio-ir-tx,gpio_pin=17
dtoverlay=gpio-ir,gpio_pin=16`

There you go

leon-anavi, 2020-03-31:
> dtoverlay=gpio-ir-tx,gpio_pin=17
> dtoverlay=gpio-ir,gpio_pin=16

Hm, this doesn't seem correct. On ANAVI Infrared pHAT the receiver is connected to GPIO 18, **not** GPIO 16 as in your settings. I will double check the user's manual in the coming days. In the mean time here is a screenshot of the board schematics in KiCad:

[image]

Best regards, Leon

SanerBCN, 2020-04-01:
I am not using Anavi to receive and send code, I was refering to the LIRC setup and the order in which the dtoverlay lines need to appear in the config.txt file.

atax112, 2020-07-03:
> Hm, this doesn't seem correct. On ANAVI Infrared pHAT the receiver is connected to GPIO 18, **not** GPIO 16 as in your settings. I will double check the user's manual in the coming days. In the mean [...]

Hello.
Registered to reply.

I have the ANAVI IR pHAT for two weeks, havent found much time to make the AC remote.

Today I continued with the setup but ran into the `hardware does not support sending
Error running command: Input/output error` issue.

Googled around quite som time, read some thread until I finally found @SanerBCN comment about the pin order.

**tx pin needs to be declared FIRST!!!**

This was it, I updated the config and it just works.
Running on ROS Lite, lates version.

Followed [](https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-infrared-phat/anavi-infrared-phat.md#setting-up-lirc) for basic setup and LIRC install.

After, I have followed [](https://www.cnx-software.com/2017/03/12/how-to-control-your-air-conditioner-with-raspberry-pi-board-and-anavi-infrared-phat/) to record my Panasonic AC remote codes. Much more straight forward than the included LIRC steps.

Heureka! I just turned my AC on and than off with two commands :)

Thank you and happy DIY.

leon-anavi, 2020-07-03:
Hi @atax112,

Thank you for using ANAVI Infrared pHAT and for the valuable feedback regarding ROS Lite. The user's manual is for Raspbian. I will have a gitve ROS Lite a try in the coming weeks.

Please feel free to suggest an edit of the user's manual (or even submit a GitHub pull request) to make it easier for other user who may run in a similar situation.

Thanks,
Leon

atax112, 2020-07-03:
> Hi @atax112,
>
> Thank you for using ANAVI Infrared pHAT and for the valuable feedback regarding ROS Lite. The user's manual is for Raspbian. I will have a gitve ROS Lite a try in the coming weeks. [...]

Hi Leon.

I have made my own step by step guide to my specific setup, not sure how to contribute it to this great project of yours.
Its a clean install with ROS Lite.

I am new to git, github and not sure how to edit or what a pull request is, sorry :)

leon-anavi, 2020-07-03:
Hi @atax112,

OK. If you want you can share the steps here as a comment. I will go through them to test it and add them as a separate ROS Lite chapter in the user's manual.

Thanks,
Leon

atax112, 2020-07-03:
> 
> 
> Hi @atax112,
> 
> OK. If you want you can share the steps here as a comment. I will go through them to test it and add them as a separate ROS Lite chapter in the user's manual.
> 
> Thanks,
> Leon

Hi @leon-anavi 

Here is my step by step approach for a clean install which I am using now, I am updating the guide because I mean to install openhab on top.

Thank you.

-----------------------------------------------

Rpi with ANAVI Infrared pHAT clean setup:

Clean 8GB SDCARD

Download Rpi imager from:
https://www.raspberrypi.org/downloads/

Run imager, select RpiOS Lite

--------------------------------------------

Enable WiFi:

Create a wpa_supplicant.conf file with following details and copy it to the boot/root folder on the SDCARD

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country= "Enter two letter country code" SK for Slovakia, GB, US, etc.

network={
 ssid=""
 psk=""
}

--------------------------------------------

Setup SSH:

Create an empty ssh file in the SDCARD boot/root folder

Just ssh, no extension, no content

--------------------------------------------

First login:
hostname: raspberrypi
user:pi
password:raspberry

Use terminal program, putty, securecrt, cmd, etc...

Change password for user pi:
sudo passwd pi

--------------------------------------------

ANAVI pHAT and LIRC setup:

sudo apt-get update
sudo apt-get install -y git i2c-tools vim
sudo raspi-config
	Interfaces
	enable I2C
[... 140 lines cut ...]
----------------------

Open a new file on the Rpi:

sudo nano lircd-panasonic-ac.conf
	copy contents from remote config txt

Replace original LIRC config with recorded remote config:
sudo cp lircd-panasonic-ac.conf /etc/lirc/lircd.conf

sudo reboot

	Optional:
	sudo systemctl restart lirc
	sudo systemctl reload lirc

To list configured commands:
irsend LIST panasonic ""

To send commands:
irsend SEND_ONCE panasonic "command"

-------------------------------------

OpenHAB install not yet finished.

atax112, 2020-07-04:
Hi @leon-anavi . 

Not sure how to contact you, i failed googling a guide to setup the infrared pHAT in openHAB.

Currently I have it working through ssh, but would like to be able to control my AC from phone or even outside local network.

I have found your guide for the temperature and humidity sensors, however nothing for the IR, not even third party guides.
Ive got openhab up and running and the java transformation with MQTT installed as well.

Could you please point me towards some instructions how to set the infrared pHAT up with openhab?

Many thanks.

leon-anavi, 2020-07-04:
Hi @atax112,

For any inquiries you can send an email to info at anavi.technology. For technical question it is also fine to open a new issue in GitHub.

As for OpenHab, please have a look at [Exec Binding](https://www.openhab.org/addons/bindings/exec/). With it you can execute command or a script. If ANAVI Infrared pHAT is on different Raspberry Pi than the one on which OpenHAB is working you should create a script with SSH for remote logic and execute it again with the Exec Binding.

Best regards,
Leon

atax112, 2020-07-04:
> Hi @atax112,
>
> For any inquiries you can send an email to info at anavi.technology. For technical question it is also fine to open a new issue in GitHub.
>
> As for OpenHab, please have a look at [Exec [...]

Thank you for your reply.

I have the pHAT on the same Rpi as the openhab server is running on, however I couldnt figure out the exec binding either.

I dont seem to understand the basics, how many steps need to be done for a simple "turn AC on" button. 

I am now investigating a way to have a shortcut on my phone which sends the command directly to the Rpi via ssh instead.

Im sure im not the only one trying to do something similar, but somehow I am not able to find a simple straight forward tutorial.

voquockha1997, 2021-04-16:
Hello @ leon-anavi
I have already tested my IR remote shield in a Raspberry PI 3B and works ok (with Raspbian + Lirc). 
Now I want to make it run in my raspberry pi 3 with Ubuntu Server 18.04, Here is my uname -a :
Linux ubuntu 4.15.0-1031-raspi2 #33-Ubuntu SMP PREEMPT Wed Jan 16 09:52:45 UTC 2019 aarch64 aarch64 aarch64 GNU/Linux
I tried with lirc and gpio-ir but neither of both works! Can you send me a suggestion ? Thanks a lot!
Regards!