AnaviTechnology/anavi-docs #21: transmission failed Error running command: Input/output error.

Carccha, 2019-11-07:
Hi, 
I'm on Raspbian Buster. I checked and applied all of the recommandations.
I can record remotes but when I test to send a IR Signal I've got -> transmission failed
Error running command: Input/output error.

/boot/config.txt:
dtoverlay=gpio-ir,gpio_pin=18
dtoverlay=gpio-ir-tx,gpio_pin=17

/etc/lirc/lirc_options.conf
driver          = default
device          = /dev/lirc0

Have you any idea ?

leon-anavi, 2019-11-07:
Hi @Carccha,

The configurations seems fine. After changing device to /dev/lirc0 have you started/restarted the systemd service for LIRC?

```
sudo systemctl start lircd
```
or:
```
sudo systemctl restart lircd
```

Best regards,
Leon

Carccha, 2019-11-08:
Hi Leon,

Yes, I followed step by step the instructions in the pdf update.
I precise, I've got this issue with a new installation of Raspbian Buster

I will retry this evening.

Kind Regards.

Carccha, 2019-11-12:
Hi Leon, 

I reinstall from scratch my RPI V3 with Raspbian Buster.
I have to install python3-setuptools in addition of all packages.

When I send: `irsend LIST tv ""`
I've got: 0000000000000000 KEY_POWER

And When I send: `irsend SEND_ONCE tv KEY_POWER`
I've always got: 
                      unknown remote: "hifi"
                      Error running command: Input/output error

leon-anavi, 2019-11-12:
Hi @Carccha,

Please note that in section **Setting up LIRC** of the user's manual **python3-setuptools** is part of the packages that have to be installed during the first step for the chapter **Install dependencies**. It is actually the last package in the long list of dependencies.

Having 0000000000000000 in the lircd.conf means that the scanning of the remote control was not successful. Please scan the remote control again with LIRC by following the steps in section **Using IR LED**: https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-infrared-phat/anavi-infrared-phat.md#using-ir-led

Best regards,
Leon

Carccha, 2019-11-12:
I redo the procedure with the same issues.

In syslog, here we've got:

Nov 12 11:05:01 raspberrypi lircd[1530]: lircd-0.10.1[1530]: Notice: accepted new client on /var/run/lirc/lircd
Nov 12 11:05:01 raspberrypi lircd-0.10.1[1530]: Notice: accepted new client on /var/run/lirc/lircd
Nov 12 11:05:01 raspberrypi lircd[1530]: lircd-0.10.1[1530]: **Info: Cannot configure the rc device for /dev/lirc0**
Nov 12 11:05:01 raspberrypi lircd[1530]: lircd-0.10.1[1530]: Error: invalid send buffer
Nov 12 11:05:01 raspberrypi lircd[1530]: lircd-0.10.1[1530]: Error: this remote configuration cannot be used to transmit
Nov 12 11:05:01 raspberrypi lircd[1530]: lircd-0.10.1[1530]: Error: error processing command: SEND_ONCE tv KEY_POWER
Nov 12 11:05:01 raspberrypi lircd[1530]: lircd-0.10.1[1530]: Error: transmission failed
Nov 12 11:05:01 raspberrypi lircd[1530]: lircd-0.10.1[1530]: Info: removed client
Nov 12 11:05:01 raspberrypi lircd-0.10.1[1530]: Info: Cannot configure the rc device for /dev/lirc0
Nov 12 11:05:01 raspberrypi irsend: lirc_command_run: Sending: SEND_ONCE tv KEY_POWER
Nov 12 11:05:01 raspberrypi lircd-0.10.1[1530]: Error: invalid send buffer
Nov 12 11:05:01 raspberrypi lircd-0.10.1[1530]: Error: this remote configuration cannot be used to transmit
Nov 12 11:05:01 raspberrypi lircd-0.10.1[1530]: Error: error processing command: SEND_ONCE tv KEY_POWER
Nov 12 11:05:01 raspberrypi lircd-0.10.1[1530]: Error: transmission failed
Nov 12 11:05:01 raspberrypi irsend: lirc_command_run, state: 0, input: "BEGIN"
Nov 12 11:05:01 raspberrypi irsend: lirc_command_run, state: 1, input: "SEND_ONCE tv KEY_POWER"
Nov 12 11:05:01 raspberrypi irsend: lirc_command_run, state: 2, input: "ERROR"
Nov 12 11:05:01 raspberrypi irsend: irsend: command failed: SEND_ONCE tv KEY_POWER
Nov 12 11:05:01 raspberrypi irsend: lirc_command_run, state: 3, input: "DATA"
Nov 12 11:05:01 raspberrypi irsend: lirc_command_run, state: 4, input: "1"
Nov 12 11:05:01 raspberrypi irsend: lirc_command_run, state: 5, input: "transmission failed"
Nov 12 11:05:01 raspberrypi irsend: lirc_command_run, state: 6, input: "END"
Nov 12 11:05:01 raspberrypi irsend: lirc_command_run: status:END, status:5
Nov 12 11:05:01 raspberrypi lircd-0.10.1[1530]: Info: removed client

leon-anavi, 2019-11-12:
According to the log "this remote configuration cannot be used to transmit", probably because of the 0000000000000000 value you mentioned in your previous post. This is **not** an OK value for a button so please switch to /dev/lirc1 and scan again the remote control with irrecord. You need to have properly scanned remote control to proceed successfully with irsend.

Best regards,
Leon

Carccha, 2019-11-12:
Leon, 

After some research [here](https://gist.github.com/prasanthj/c15a5298eb682bde34961c322c95378b)

I add the following lines to /etc/modules file
lirc_dev
lirc_rpi gpio_in_pin=18 gpio_out_pin=17

And it's finally works.
Maybe that helped someone.

Thanks for your help and your time.
Kind regards.

closed 2019-11-12 by Carccha

leon-anavi, 2019-11-12:
Hi @Carccha,

Thank you for the feedback. I have a look at the user's manual and add your feedback about  /etc/modules to it.

Thanks,
Leon