AnaviTechnology/anavi-docs #22: Simple remote LED candle with IR HAT

AndrewD93, 2019-11-12:
So I've bouht this nice remote-controlled LED candle and the first thing I thought once home was trying to map the two simple buttons right into the IR HAT.
[image]
I mapped them as raw signals since this worked fine with my air conditioner and the signals appear to be pretty basic. When I try it with irsend, the signal is correctly sent but the candle won't turn on (I've tried an air conditioner command right before and it works fine). [Here's my config file.](https://github.com/AnaviTechnology/anavi-docs/files/3838074/candle.lircd.conf.txt)
Do you have any clues on what it might be?
(I'm sorry if this comes out to be off-topic, I remember you assisted me finely months ago)

leon-anavi, 2019-11-12:
Hi @AndrewD93,

Cool candle! I love it. Never used such a LED candle but I may buy one too :) 

Unlike the remote controls for air conditioners, this remote looks simple. Have you tried scanning it with irrecond as standard codes (I mean not raw codes)?

Best regards,
Leon

leon-anavi, 2019-11-12:
@AndrewD93 one more thing... (totally off-topic): right now Crowd Supply is running a field report contest. st. If you share a few words (before 15 November) about home automation of air conditioner with ANAVI Infrared pHAT tou will get a $25 Crowd Supply credit and you will enter a raffle for more cool open source gear. Details how to participate are available here:
https://www.crowdsupply.com/anavi-technology/infrared-phat/updates/anavi-field-report-contest

It will be cool to share your experience with the complex remote control for air conditioner and the the raw codes in LIRC.

Thanks, Leon

AndrewD93, 2019-11-12:
So when I try to do it with non-raw signals it takes horribly long to make initial calibration with dots and I get these warnings during the procedure.
`Cannot find any gap, using an arbitrary 50 ms one. If you have a regular remote for e. g., a TV or such this is probably a point where you hit control-C. However, technical hardware like air                                                           condition gear often works without any gap. If you think it's reasonable that your remote lacks gap you can proceed.`

`Checking for toggle bit mask. Please press an arbitrary button repeatedly as fast as possible.                                                        Make sure you keep pressing the SAME button and that you DON'T HOLD the button down!. If you can't see any dots appear, wait a bit between button presses.`

[The output remote file](https://github.com/AnaviTechnology/anavi-docs/files/3838381/candled.lircd.conf.txt) is basically all zeros and when I try to irsend it I get a "transmission failed" error.

I'll take a better look at your contest and will be happy to take part :)

leon-anavi, 2019-11-12:
Earlier today @Carccha reported a similar issue on Raspbian Buster. He solved it by additional configuration in **/etc/modules**. This has not been yet added to the user's manual (but it has to go there). Here are the details: https://github.com/AnaviTechnology/anavi-docs/issues/21#issuecomment-552838471 

Is it possible that you are facing something similar? Can you please give it a try?

Thanks,
Leon

AndrewD93, 2019-11-12:
Well my configuration file is already like that and I think you've already updated user's manual with that. That's the issue I experienced this summer. 
I think that error occurs because of how the standard config file right above which is basically made of zeros. In fact, if I try any different remote it works properly (execpt the candle not turning on, of course).

leon-anavi, 2019-11-12:
Hm, sorry, I am running out of ideas. Indeed the zeros are not a good sing. May be there is some kind of incompatibility with this remote.

On the off-topic - I am excited that you are interested in taking part in the Crowd Supply contest!

Best regards,
Leon