AnaviTechnology/anavi-docs #23: LIRC installation problem

Mustafa744, 2020-01-28:
i have raspberry pi 3 runing raspian buster .
i followed the instructions  in 
[https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-infrared-phat/anavi-infrared-phat.md](url) 
but when i type `patch -p0 -i lirc-gpio-ir-0.10.patch
i get  : `
pi@raspberrypi:~/lirc-src $ patch -p0 -i lirc-gpio-ir-0.10.patch
can't find file to patch at input line 4
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff -ruN lirc-0.10.1.orig/lib/config_file.c lirc-0.10.1/lib/config_file.c
|--- lirc-0.10.1.orig/lib/config_file.c 2017-09-10 17:52:19.000000000 +0900
|+++ lirc-0.10.1/lib/config_file.c      2019-06-26 00:39:45.734320696 +0900
--------------------------
File to patch:
`
how can i fix this problem?

leon-anavi, 2020-01-29:
Hi @Mustafa744,

Seems you are executing the patch command in the directory where the source of LIRC is. Please follow all exact steps from user's manual about setting LIRC:
https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-infrared-phat/anavi-infrared-phat.md#setting-up-lirc

I've just tested the part with the patch and everything seems fine therefore I will close the issue. Please fee free to open it again if you have any follow-up questions. Here is the log:

```
pi@raspberrypi:~ $ mkdir ~/lirc-src
pi@raspberrypi:~ $ cd ~/lirc-src
pi@raspberrypi:~/lirc-src $ apt source lirc
Reading package lists... Done
NOTICE: 'lirc' packaging is maintained in the 'Git' version control system at:
https://gitlab.com/leamas/lirc.git
Please use:
git clone https://gitlab.com/leamas/lirc.git
to retrieve the latest (possibly unreleased) updates to the package.
Need to get 2,754 kB of source archives.
Get:1 http://mirrors.linux-bg.org/raspbian/raspbian buster/main lirc 0.10.1-5.2 (dsc) [2,663 B]
Get:2 http://mirrors.linux-bg.org/raspbian/raspbian buster/main lirc 0.10.1-5.2 (tar) [2,715 kB]
Get:3 http://mirrors.linux-bg.org/raspbian/raspbian buster/main lirc 0.10.1-5.2 (diff) [35.9 kB]
Fetched 2,754 kB in 3s (1,080 kB/s)
dpkg-source: info: extracting lirc in lirc-0.10.1
dpkg-source: info: unpacking lirc_0.10.1.orig.tar.gz
dpkg-source: info: unpacking lirc_0.10.1-5.2.debian.tar.xz
dpkg-source: info: using patch list from debian/patches/series
dpkg-source: info: applying 0001-lirc.org-Remove-non-free-advertising.patch
dpkg-source: info: applying 0002-lirc-setup-Fix-crash-on-start-on-missing-lirc.config.patch
dpkg-source: info: applying 0003-logging-Don-t-use-broken-LOG_CONS-syslog-flag.patch
dpkg-source: info: applying 0004-lircd-Fix-connect-option-parsing-error-343.patch
dpkg-source: info: applying 0005-systemd-support-Notify-systemd-on-successful-startup.patch
pi@raspberrypi:~/lirc-src $ wget https://raw.githubusercontent.com/neuralassembly/raspi/master/lirc-gpio-ir-0.10.patch
--2020-01-29 02:18:26--  https://raw.githubusercontent.com/neuralassembly/raspi/master/lirc-gpio-ir-0.10.patch
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.0.133, 151.101.64.133, 151.101.128.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.0.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 5223 (5.1K) [text/plain]
Saving to: ‘lirc-gpio-ir-0.10.patch’

lirc-gpio-ir-0.10.patch      100%[==============================================>]   5.10K  --.-KB/s    in 0.001s  

2020-01-29 02:18:27 (8.70 MB/s) - ‘lirc-gpio-ir-0.10.patch’ saved [5223/5223]

pi@raspberrypi:~/lirc-src $ patch -p0 -i lirc-gpio-ir-0.10.patch
patching file lirc-0.10.1/lib/config_file.c
patching file lirc-0.10.1/lib/ir_remote.h
patching file lirc-0.10.1/lib/irrecord.c
patching file lirc-0.10.1/lib/lirc/ir_remote.h
patching file lirc-0.10.1/tools/mode2.cpp
```

Best regards,
Leon

Mustafa744, 2020-01-29:
@leon-anavi  the problem is that when i run >>apt source lirc
i get lirc version 0.9.4c (instead of 0.10.1) despite updating my raspberry pi using `sudo apt update`
how can i solve this problem?

leon-anavi, 2020-01-29:
Hi @Mustafa744,

Hm, 0.9.4c  is the old version. Could you please open a terminal, type in `lsb_release -a` and share the output here?

Thanks,
Leon

Mustafa744, 2020-01-29:
`pi@raspberrypi:~/lirc-src $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.11 (stretch)
Release:        9.11
Codename:       stretch
`

leon-anavi, 2020-01-29:
@Mustafa744, you are using Raspbian Stretch. It is recommended to upgrade to Raspbian Buster.

When you have buster the output will be:
```
pi@raspberrypi:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
```

[The latest user's manual of ANAVI Infrared pHAT is for Raspbian Buster](https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-infrared-phat/anavi-infrared-phat.md). Details are available here: https://blog.anavi.technology/?p=1

Although it is recommended to upgrade to Raspbian Buster, it is also possible to use ANAVI Infrared pHAT with Raspbian Stretch following [the old user's manual](https://github.com/AnaviTechnology/anavi-docs/blob/4ad760b2dbc76ff28e9732e789f66a42788e9692/anavi-infrared-phat/anavi-infrared-phat.md). There are significant differences between Raspbian Stretch and Buster in terms of infrared.

Best regards,
Leon

Mustafa744, 2020-01-29:
@leon-anavi ok i will use the old manual's instructions.
thanks for your help ❤

leon-anavi, 2020-01-29:
ok, you are welcome. Keep in mind that in future it is highly recommended to upgrade to Buster.

Best regards,
Leon

closed 2020-01-29 by leon-anavi