AnaviTechnology/anavi-docs #27: [ANAVI Infrared pHAT] How to access a sensor on UART?

craftzdog, 2020-04-15:
Hi. Thanks for the answer to [my previous question](https://github.com/AnaviTechnology/anavi-gas-detector-sw/issues/2).
Could I ask one more?

I have an ANAVI Infrared pHAT (Advanced Kit) for my Raspberry Pi 3.
It works great to get temperature, humidity, etc. I'm totally happy with it!
I'd like to also visualize CO2 level and bought a MH-Z19B sensor which supports UART.
I successfully got it to work when directly connected to my Rasp via GPIO (`/dev/serial0`) with [a Python module](https://github.com/UedaTakeyuki/mh-z19) like so:

[image]

```
pi@raspberrypi:~ $ gpio readall
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 | ALT0 | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 | ALT0 | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 1 | ALT5 | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | ALT5 | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI |   IN | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO |   IN | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK |   IN | 0 | 23 || 24 | 1 | IN   | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | IN   | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+

pi@raspberrypi:~ $ sudo python -m mh_z19
{"co2": 1917}
```

But when I connect it with the UART of the Infrared pHAT, it doesn't work.
Is it possible to do so?
Here is [the manual for MH-Z19B](https://www.winsen-sensor.com/d/files/infrared-gas-sensor/mh-z19b-co2-ver1_0.pdf).
Thanks in advance!

craftzdog, 2020-04-15:
Got it to work!
I figured out that I have to connect pins like this:

```
Sensor - pHAT UART
RX - TX
TX - RX
```

Now it works great.
Sorry for disturbing and thank you for creating the great device!

closed 2020-04-15 by craftzdog

leon-anavi, 2020-04-15:
Hi @craftzdog,

Thank you for the feedback about the product. I am glad to hear that you figured out the proper wiring of the module and you enjoy Infrared pHAT.

I am using this opportunity to share some technical details that might be useful during the development of more advanced projects like the one you are working on. Following Raspberry Pi Foundation guidelines for Hardware Attached on Top (HAT), ANAVI Infrared pHAT has an EEPROM connected to the secondary I2C interface on Raspberry Pi pins 27 and 28. The EEPROM stores device tree binary overlay configurations that set the state of all pins on the header for the specific needs of ANAVI Infrared pHAT. As you have already done, after booting they can be obtained from `gpio readall`.

Best regards, Leon