AnaviTechnology/anavi-docs #28: irrecord 0x0

MargaritaZheng, 2020-04-19:
I am using the ANAVI ir recorder on Buster on my Raspberry Pi 4. I have installed all patches as detailed by the installation guides in this repo. 

I am able to complete the irrecord process, however, all commands are saved as 0x0. I have tried with 3 different IR remotes.  

Please help me, I have been trying this for days.

My /etc/lirc/lirc_options.conf:
[lircd]
nodaemon        = False
driver          = default
device          = /dev/lirc1
output          = /var/run/lirc/lircd
pidfile         = /var/run/lirc/lircd.pid
plugindir       = /usr/lib/arm-linux-gnueabihf/lirc/plugins
permission      = 666
allow-simulate  = No
repeat-max      = 600

Here is my config file:

/boot/config.txt
# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=off
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
dtoverlay=lirc-rpi
#dtoverlay=lirc-rpi,gpio-in_pin=18,gpio_out_pin=17
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18
#dtoverlay=gpio-ir,gpio_out_pin=17,gpio_in_pin=18,gpio_in_pull=up
# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
#audio_pwm_mode = 2

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
#dtoverlay=lirc-rpi,gpio_in_pin=18,gpio_out_pin=17
# NOOBS Auto-generated Settings:
hdmi_force_hotplug=1
dtparam=spi=on

#dtoverlay=lirc-rpi,gpio_in_pin=18,gpio_out_pin=17
dtoverlay=gpio-ir,gpio_pin=18

leon-anavi, 2020-04-19:
Hi,

Thank you for using ANAVI Infrared pHAT. The steps for installing and configuring Raspbian Buster have been described in [the user's manual section **Infrared and LIRC**](https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-infrared-phat/anavi-infrared-phat.md#infrared-and-lirc).

I will try to help you debug the issue you are having. Could you please open a terminal, type in the following commands and share the output?

* Check Raspbian version:
```
lsb_release -a
```
* Check LIRC version:
```
lircd -v
```

Best regards,
Leon

MargaritaZheng, 2020-04-19:
Hi Leon,

Thank you so much for your quick response!  Here are my outputs.

pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster
pi@raspberrypi:~ $ lircd -v
lircd 0.10.1

leon-anavi, 2020-04-19:
Hi @MargaritaZheng,

It seems you are using the right versions. However, as far as I can see your configuration files are not correct. Please fix the following [the user's manual](https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-infrared-phat/anavi-infrared-phat.md#infrared-and-lirc). Some pointers:

* For scanning a remote control in `/etc/lirc/lirc_options.conf` set device to `/dev/lirc1`:

```
driver          = default
device          = /dev/lirc1
```

Device `/dev/lirc1` is the receiver and device `/dev/lirc0` is the transmitter. Initially `/dev/lirc1` is used to scan a remote control. After that the configuration has to be updated to `/dev/lirc0` in order to send infrared commands.

* In `/boot/config.txt`:

```
dtoverlay=gpio-ir,gpio_pin=18
dtoverlay=gpio-ir-tx,gpio_pin=17
```

Please note the exact pins. You can just copy & paste these lines. The line must **not** be commented, aka it should start without `#`. The line for `lirc-rpi` that you have right now should be removed.

Best regards,
Leon

MargaritaZheng, 2020-04-19:
I apologize, I think I did not copy and paste the entire config file. I had
those two lines

dtoverlay=gpio-ir,gpio_pin=18
dtoverlay=gpio-ir-tx,gpio_pin=17

at the bottom of the config file and they did not copy paste over when
I posted my issue. I had followed the installation instructions from
your github and had made those changes already.

leon-anavi, 2020-04-19:
OK, but you still need to remove `dtoverlay=lirc-rpi` from `/boot/config.txt`. In 2019, with the release Buster of Raspbian, module **lirc_rpi**, the Linux kernel module provided with Raspbian before, was replaced with **gpio-ir** and **gpio-ir-tx**. If you are interested in details have [a look here](https://blog.anavi.technology/?p=1).

Also do you have the following lines in `/etc/modules`?

```
lirc_dev
lirc_rpi gpio_in_pin=18 gpio_out_pin=17
```

Best regards,
Leon

MargaritaZheng, 2020-04-19:
Yes, I have those lines in my /etc/modules. And I had uncommented the
dtoverlay=lirc-rpi
out of desperation, but I have run it with it commented out and it still
does not record any signals.

leon-anavi, 2020-04-19:
Have you downloaded and build from source LIRC (with the required patch) as explained in [the appropriate section of the user's manual](https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-infrared-phat/anavi-infrared-phat.md#setting-up-lirc)?

Please note that you have to do all steps from the user's manual exactly. Even if a single configuration is different this may result in issues as the ones you are currently experiencing.

What output do you get when you run in  a terminal the following command and start pressing keys on the remote control?

```
mode2 -d /dev/lirc1
```

Best regards,
Leon

MargaritaZheng, 2020-04-20:
Yes, I downloaded and built from source LIRC with the patches as
explained in the manual.

When I run mode2, I get a long series of
start
pulse

numbers. They run without me pressing any keys on the remote control,
however.

leon-anavi, 2020-04-20:
Hi @MargaritaZheng,

Earlier today I've followed the steps in the user's manual and LIRC was working fine on Rapsbian Buster and Raspberry Pi 4. Command `mode2 -d /dev/lirc1` should output only when a button is pressed.

Could you please double check and share your exact `/boot/config.txt` and `/etc/lirc/lirc_options.conf`?

Best regards,
Leon

MargaritaZheng, 2020-04-22:
Thanks so much for taking a look at this Leon.

[config.txt](https://github.com/AnaviTechnology/anavi-docs/files/4514575/config.txt)
[lirc_options.txt](https://github.com/AnaviTechnology/anavi-docs/files/4514576/lirc_options.txt)

leon-anavi, 2020-04-22:
Hi @MargaritaZheng,

lirc_options.txt looks good. Your config.txt also looks ok with a couple of remarks: UART is disable and SPI is enabled. ANAVI Infrared pHAT has a UART pins for serial communication and does **not** use SPI. It is recommended to:
* Enable UART: either [through raspi-config](https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-infrared-phat/anavi-infrared-phat.md#serial-debugging) as explained in the user's manual or by manually setting `enable_uart=1` in `/boot/config.txt`
* Disable SPI:  either through raspi-config or by manually removing `dtparam=spi=on` in `/boot/config.txt`

For  a reference here is a `/boot/config.txt` known to work on Raspberry Pi 4 with Raspbian Buster:

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
enable_uart=1

dtoverlay=gpio-ir,gpio_pin=18
dtoverlay=gpio-ir-tx,gpio_pin=17
```

However, technically speaking neither UART nor SPI are connected to the LIRC features on the board.

Before running `mode2 -d /dev/lirc1` have you stopped the lircd systemd service by executing `sudo systemctl stop lircd` ?

Another explanation might be some kind of infrared light interference detected by LIRC while running `mode2 -d /dev/lirc1`.

ANAVI Infrared pHAT relies on [Vishay TSOP34838 infrared receiver module](https://www.mouser.bg/datasheet/2/427/tsop322-1767012.pdf) and all boards have been carefully manually tested before providing them to distributors and customers. So the board you have has passed QA checks, including with the `mode2` command.

Best regards,
Leon

MargaritaZheng, 2020-04-22:
Hi Leon,

I changed my config file to match the one you sent. And yes, I am running sudo systemctl stop lircd before testing with mode2. However, I still get many pulse space responses without pressing the remote button. In addition, I have tried this in different rooms of my house and I am still getting the same result. Is there anything else I can try?

leon-anavi, 2020-04-26:
Hi @MargaritaZheng,

Sorry for my slow reply. I am running out of ideas. It seems you are doing the configuration properly. I'm attaching [the deb files build from the patches LIRC source following the procedure in the user's manual](https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-infrared-phat/anavi-infrared-phat.md#setting-up-lirc):

[lirc-0.10.1-rpi-patch.zip](https://github.com/AnaviTechnology/anavi-docs/files/4536519/lirc-0.10.1-rpi-patch.zip)

You can give them a try. There are from my fresh install of the setup on rpi4. 

In general, all boards have been manually tested before providing them to distributors and customers. If you think you may be experiencing the issues due to damage during the transportation please send an email to info at anavi.technology. Right now shipping replacements is almost impossible due to logistic implications from the COVID-19 outbreak but we will figure out something...

Best regards,
Leon

MargaritaZheng, 2020-04-26:
Hi Leon,

Thank you for trying to debug this with me. I think there is something wrong with the board, possibly from shipping, as I have just run through the entire process again on a brand new raspberry pi 4 running buster, and I am getting the same results. I will send an email shortly.

Best,
Margarita

xXAzazelXx, 2020-08-19:
Hi Leon,

I appear to have this same issue also.
I am running this on old RaspberryPi B+

```
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
pi@raspberrypi:~ $ lircd -v
lircd 0.10.1
```
When running `mode2 -d /dev/lirc1` I get random space XXX and pulse XXX if I put my hand in front of IRTX1 and IRTX2.

If i point my remote at IRRX I am also getting space XXX and pulse XXX.

The issue comes with `irrecord -d /dev/lirc1 ~/lircd.conf`.  If  I was to say pick KEY_POWER value and hold the power button on my remote after I save it all I get 

```
      begin codes
         KEY_POWER             0x0
      end codes
```

Any tips?

leon-anavi, 2020-08-19:
Hi @xXAzazelXx,

Please made sure there is no interference that might be caused by radio signals or other infrared devices. Try scanning the remote control with  `--force` (or just -f) for raw mode. For example:

```
irrecord -f -d /dev/lirc1 ~/lircd.conf
```

What remote control are you trying to scan? If you are trying to scan a remote control of AC you will need a different approach as explained here (however keep in mind that the article is a bit outdated and the steps nowadays might be slightly different):
https://www.cnx-software.com/2017/03/12/how-to-control-your-air-conditioner-with-raspberry-pi-board-and-anavi-infrared-phat/

Best regards,
Leon

xXAzazelXx, 2020-08-21:
Hi @leon-anavi 

Thank you, I think I am now getting another issue, I'm happy to open a new issue if you would like.

- Started fresh with Pi0w, Buster 10
- Followed the Github/PDF guide. 
- Created _liblirc0_0.10.1-6.2~deb10u1_armhf.deb, liblircclient0_0.10.1-6.2~deb10u1_armhf.deb, lirc_0.10.1-6.2~deb10u1_armhf.deb_
PS: I didn't have to do xxx.dist configs and my install worked on the first try.
- Updated lirc_options.conf and updated /boot/config.txt
- Ran the install again with _--allow-downgrades_
PS:  I got these messages but it seems to have re-installed.
```
Note, selecting 'liblirc0' instead of './liblirc0_0.10.1-6.2~deb10u1_armhf.deb'
Note, selecting 'liblircclient0' instead of './liblircclient0_0.10.1-6.2~deb10u1_armhf.deb'
Note, selecting 'lirc' instead of './lirc_0.10.1-6.2~deb10u1_armhf.deb'
```

Anyway, after reboot, once again I am getting signals with `mode2 -d /dev/lirc1` and now irrecord saves the remote config:

```
  name  sony
  bits            4
  flags SPACE_ENC|CONST_LENGTH
  eps            30
  aeps          100

  header       2427   585
  one          1219   579
  zero          619   579
  post_data_bits  8
  post_data      0x90
  gap          45031
  toggle_bit_mask 0x0
  frequency    38000

      begin codes
          KEY_VOLUMEUP             0x4
          KEY_VOLUMEDOWN           0xC
          KEY_POWER                0xA
      end codes
```

-  I switch back to /dev/lirc0 and copy the new config across

pi@raspberrypi:~ $ irsend list sony ""

```
0000000000000004 KEY_VOLUMEUP
000000000000000c KEY_VOLUMEDOWN
000000000000000a KEY_POWER
```

Command:

`irsend send_once sony KEY_POWER`

LOGS:

```
Aug 21 03:20:23 raspberrypi lircd-0.10.1[468]: Warning: Running as root
Aug 21 03:20:23 raspberrypi lircd-0.10.1[468]: Info: Using remote: sony.
Aug 21 03:20:23 raspberrypi lircd-0.10.1[468]: Notice: lircd(default) ready, using /var/run/lirc/lircd
Aug 21 03:21:51 raspberrypi lircd-0.10.1[468]: Notice: accepted new client on /var/run/lirc/lircd
Aug 21 03:21:51 raspberrypi lircd-0.10.1[468]: Info: Cannot configure the rc device for /dev/lirc0
Aug 21 03:24:25 raspberrypi lircd-0.10.1[468]: Info: removed client
Aug 21 03:24:30 raspberrypi lircd-0.10.1[468]: Notice: accepted new client on /var/run/lirc/lircd
Aug 21 03:24:30 raspberrypi lircd-0.10.1[468]: Info: Cannot configure the rc device for /dev/lirc0
Aug 21 03:24:30 raspberrypi irsend: lirc_command_run: data:END, status:0
Aug 21 03:24:30 raspberrypi lircd-0.10.1[468]: Info: removed client

```

leon-anavi, 2020-08-21:
Hi @xXAzazelXx,

Well done for solving the issue with scanning remote controls. Could you please open a separate issue about the infrared transmission with all you configuration details?

Thanks,
Leon