AnaviTechnology/anavi-docs #29: Looking for information about miracle controller mqtt control

xaque208, 2020-04-22:
Hello, just received the controller and very excited to integrate it into my system.

I've connected the controller to a mosquitto server running in my basement, and watching all the messages being sent with `mosquitto_sub`.  I'm wondering how to manage the `/config` endpoint.  Is this a protocol invented by Anavi, or a community standard?  Either way, would you mind pointing me at some reading material so I can build something to parse those messages correctly and know how to use them.  In this case, I suppose I'd want to change things about the lights and such, but I'm not using home assistant or other platform.

leon-anavi, 2020-04-22:
Hi @xaque208,

The open source firmware of ANAVI Miracle Controller follows the Home Assistant community standards with JSON payload of the following MQTT entities:

* [MQTT Light ](https://www.home-assistant.io/integrations/light.mqtt/) - for each of the LED strips

* [MQTT Sensor](https://www.home-assistant.io/integrations/sensor.mqtt/) - for any of the supported I2C sensor attached to ANAVI Miracle Controller

* [MQTT Discovery](https://www.home-assistant.io/docs/mqtt/discovery/) - for automatic discovery of the device by Home Assistant

All these specifications for JSON serialized data in the payload defined by Home Assistant have been already partially or fully adopted by other open source IoT platforms like OpenHAB and ioBroker.

It is also possible to use them directly for your custom application, for example[ this simple HTML5 web page](https://github.com/AnaviTechnology/anavi-light-demo) for demo purposes uses the Paho MQTT JavaScript client library has support for ANAVI Miracle Controller. [The live demo](http://demo.anavi.technology/) is available here and you see [how it works from user's perspective in this video](https://www.youtube.com/watch?v=zSkBg8KF_-o).

Best regards, Leon

xaque208, 2020-04-24:
Excellent, thank you for the pointers.  That's just what I was after. Cheers.

closed 2020-04-24 by leon-anavi