AnaviTechnology/anavi-docs #34: LIRC patch for Raspbian unsuccessful

latomus, 2021-04-26:
Hi Leon,
I tried to apply the guidelines described here (https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-infrared-phat/anavi-infrared-phat.md#chapter-3-software).

When executing this commande "sudo apt install -y ./liblirc0_0.10.1-6.2~deb10u1_armhf.deb ./liblircclient0_0.10.1-6.2~deb10u1_armhf.deb ./lirc_0.10.1-6.2~deb10u1_armhf.deb", I received the following error messages.

pi@RPI-LIRC-RX:~/lirc-src/lirc-0.10.1 $ cd ~/lirc-src
pi@RPI-LIRC-RX:~/lirc-src $ sudo apt install -y ./liblirc0_0.10.1-6.2~deb10u1_armhf.deb ./liblircclient0_0.10.1-6.2~deb10u1_armhf.deb ./lirc_0.10.1-6.2~deb10u1_armhf.deb
Reading package lists... Done
E: Unsupported file ./liblirc0_0.10.1-6.2~deb10u1_armhf.deb given on commandline
E: Unsupported file ./liblircclient0_0.10.1-6.2~deb10u1_armhf.deb given on commandline
E: Unsupported file ./lirc_0.10.1-6.2~deb10u1_armhf.deb given on commandline

I tried on a RPI 3 B + and on a RPI 4 4 megs, with the latest version of the Raspberry OS (issued on 2021-03-04), after a fresh install.

I am wondering what I did wrong, since I mostly copied/pasted the command lines from you tutorial...

Thank you very much in advance for any suggestions that could come to your mind ! 

Best regards

Pierre Masson

leon-anavi, 2021-04-26:
Hi @latomus,

Sorry for the inconvenience and thank you for reporting this issue. I've fixed it in the document. The LIRC version changed from 1.6.2 to 1.6.3 therefore the install command now should be:

```
sudo apt install -y ./liblirc0_0.10.1-6.3~deb10u1_armhf.deb ./liblircclient0_0.10.1-6.3~deb10u1_armhf.deb ./lirc_0.10.1-6.3~deb10u1_armhf.deb
```

Could you please try with the command above?

Thanks,
Leon

closed 2021-04-26 by leon-anavi

latomus, 2021-04-27:
Hi @leon-anavi,

Thank you very much for your swift reply.

I confirm that the revised version of the line command in question worked indeed successfully.

I still have some questions, that might be of interest to other users who are not advanced IT users :

1. - why do we have to patch LIRC for Raspberry ?
2. - will future versions of the module oblige us to re-run all these command lines or, on the contraty, is this patch stable, irregardless of future versions of the Raspberry OS ?

Best regards

Pierre

leon-anavi, 2021-04-27:
Hi Pierre,

[In 2019](https://blog.anavi.technology/?p=1) lirc_rpi, the Linux kernel module provided with Raspbian before, was replaced with gpio-ir and gpio-ir-tx. Because of this the old way for setting up ANAVI Infrared pHAT was no longer working and LIRC required this manual patch which is from the posts by [neuralassembly in the this Raspberry Pi forum thread](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=235256).

This is a known issue and there is [an open ticket in sourceforge about LIRC client with Raspberry Pi gpio kernel driver](https://sourceforge.net/p/lirc/tickets/359/). LIRC is a very complex project with a lot of backward compatibility legacy for various devices. I am not sure if this patch that works perfectly well on Raspberry Pi will have any impact on other hardware. I am also not sure what are the plans of LIRC maintainers for it. I will happy to have a straight-forward installation of LIRC on Raspberry Pi OS but I am not sure when/if it is going to be fixed in future versions of the package as this is something out of my control.

Best regards, Leon

latomus, 2021-04-27:
I had spotted all these elements while reviewing available information sources and posts on the various relevant websites. I had just not connected them into a comprehensive set of information so far and your  last reply is very enlightning in this respect.
Thanks a lot.
Best regards
P.

jkkataja, 2022-07-07:
Hi,

old issue but figured I'd rather comment here (and reopen it) than create a new one. After all, found this via Google myself also.

Just rebuild Pi4 with latest Pi OS lite 64bit

`Raspberry Pi OS Lite
Release date: April 4th 2022
System: 64-bit
Kernel version: 5.15
Debian version: 11 (bullseye)`

and noticed that I had to use the following command:
`sudo apt install -y ./liblirc0_0.10.1-6.3_arm64.deb ./liblircclient0_0.10.1-6.3_arm64.deb ./lirc_0.10.1-6.3_arm64.deb`

Also in lirc_options.conf the plugindir needs to be following `plugindir       = /usr/lib/aarch64-linux-gnu/lirc/plugins` but that most likely isn't an issue to a lot of people, I faced that since I had old options conf in my Ansible.

Cheers,
Jere

nohajc, 2024-02-06:
Hi, is it possible the lirc patch is no longer necessary?
According to the ticket (which has been closed last year), a similar patch has been applied.
https://sourceforge.net/p/lirc/tickets/359/

leon-anavi, 2024-02-06:
> Hi, is it possible the lirc patch is no longer necessary? According to the ticket (which has been closed last year), a similar patch has been applied. https://sourceforge.net/p/lirc/tickets/359/

Yes, it is possible if Raspberry Pi OS nowadays provides LIRC version with the mentioned patch. I will have to check.

commit 7c778728 (2021-04-26): anavi-infrared-phat.md: Fix LIRC installation

Change version from 1.6.2 to 1.6.3.

Fix #34

Reported-by: Pierre Masson @latomus