AnaviTechnology/anavi-docs #36: Macro pad 2 document points to video for Macro pad 8

gregkh, 2021-09-22:
Couldn't find the macro pad 2 video online anywhere for how to build it, the video points to the macro pad 8.

Also the firmware for the 2 pad keyboard does not seem to be upstream anywhere, is there a link to it somewhere online?

leon-anavi, 2021-09-22:
Hi @gregkh,

Please accept my apologizes for the inconvenience.

> Couldn't find the macro pad 2 video online anywhere for how to build it, the video points to the macro pad 8.

Sorry, I forgot to publish the assembly video. I have the raw video so I will edit and publish in the coming days. After that I'll ask Crowd Supply to spread the word about it in an update.

In the mean time, text guidelines for the assembly [are available at chapter 2 of the user's manual](https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-macro-pad-2/anavi-macro-pad-2.md#assembly).

> Also the firmware for the 2 pad keyboard does not seem to be upstream anywhere, is there a link to it somewhere online?

Unfortunately, [recently the support for all keyboards using Microchip ATtiny85, including ANAVI Macro Pad 2, was dropped from QMK](https://github.com/qmk/qmk_firmware/commit/0dafd83f72ba2ae79c2ca754a120b1de8b4dbfe8) due to the increased size of the firmware following [the implementation of a features to process combos earlier and overlapping combos](https://github.com/qmk/qmk_firmware/pull/8591). Therefore support for ANAVI Macro Pad 2 is available in older versions, for example [QMK firmware version 0.13.19](https://github.com/qmk/qmk_firmware/tree/0.13.19/keyboards/anavi/macropad2).

Thanks,
Leon

closed 2021-09-22 by gregkh

leon-anavi, 2021-09-22:
Sorry again for the inconvenience. A short video with the assembly guidelines for the developer kit [is now available at YouTube](https://www.youtube.com/watch?v=gZUdU3ZXAq8).

Thanks,
Leon