AnaviTechnology/anavi-docs #45: Error running make on BMP180/c

gtodd876, 2022-02-17:
I was able to compile BH1750 & HTU21D but I'm having an issue with BMP180

Running:
`anavi-examples/sensors/BMP180/c $ make`
Produces this error:
[image]

I'm using:
 * i2c-tools 4.2-1+b1
 * python3-rpi.gpio 0.7.0-0.2+b1
 * wiringpi 2.52
 
 Thank you in advance if anybody can help point me in the right direction!

leon-anavi, 2022-02-17:
Hi @gtodd876,

Thank you for reporting this issue and sorry for the inconvenience. After a lot of turmoil recently [library wiringpi was deprecated (probably for good)](http://wiringpi.com/wiringpi-deprecated/) in latest releases of Raspberry Pi OS this affects some of the examples. We will need to move away from wiringpi. Let's keep this issue open until it is done.

Best regards,
Leon

gtodd876, 2022-02-17:
No problem at all! Sounds great. Thank you for the information and I will research alternatives in the meantime.

gtodd876, 2022-02-19:
So far I've dug up these possible alternatives:
* [BCM2835](https://www.airspayce.com/mikem/bcm2835/)
* [Pigpio](https://abyz.me.uk/rpi/pigpio/)
* [Unofficial Mirror/Prot of WiringPi](https://github.com/WiringPi/WiringPi)

leon-anavi, 2022-05-08:
Hi @gtodd876,

I have added a Python3 script as an example for reading temperature and barometric pressure from BMP180 using smbus:
https://github.com/AnaviTechnology/anavi-examples/blob/master/sensors/BMP180/python/BMP180.py

Best regards,
Leon

gtodd876, 2022-05-10:
Thank you so much Leon!! Can't wait to use this!

closed 2022-05-10 by gtodd876