AnaviTechnology/anavi-docs #46: Configuring ANAVI Gardening Tools example set-up errors

samjacoby, 2022-03-10:
When following the directions here, https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-gardening-uhat/anavi-gardening-uhat.md#examples, apt spits out the following.

```
sudo apt install -y git git-core vim python-dev python-rpi.gpio  wiringpi i2c-tools
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'git' instead of 'git-core'
Note, selecting 'python-dev-is-python2' instead of 'python-dev'
Package wiringpi is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package python-rpi.gpio is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python-rpi.gpio' has no installation candidate
E: Package 'wiringpi' has no installation candidate
```

This is on a brand new pi install:
```
pi@hedy:~ $ uname -a
Linux hedy 5.10.92-v7+ #1514 SMP Mon Jan 17 17:36:39 GMT 2022 armv7l GNU/Linux

```
pi@hedy:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

leon-anavi, 2022-05-08:
Hi @samjacoby,

Sorry for the very late reply. Some dependencies are now deprecated. I am in the process of updating the Python and C examples. Actually the Python 3 examples are ready. The documentation will be updated accordingly soon. For C examples I have already replaced the deprecated `wiringPi` library with `libi2c-dev` for BH1750. HTU21D and BMP180 are still in the TODO list.

Here are the recent changes in the source code for the examples:
https://github.com/AnaviTechnology/anavi-examples

Best regards,
Leon

leon-anavi, 2022-05-08:
ANAVI Gardening uHAT user's manual has been updated with commit 1c8582d908e32ab1af008a38491c625cd34f22ef to use the new version of the examples.

Best regards,
Leon