AnaviTechnology/anavi-docs #47: Moisture sensor showing 100%

fpy-homeb, 2022-05-13:
I have installed following instructions 
When running example script provided with ether 1 or 2 sensors attached, the result returns 100% moisture 
This is on a pi zero w2.

I will re-run from a fresh install of bullseye to see if anything changes (this will be the 2nd time around)

Has anyone else had this issue?

leon-anavi, 2022-05-13:
Hi @fpy-homeb,

Could you please share the following details:

* Is the soil moisture sensor properly connected to the dedicate 3pin male header? Could you please share a photo? GND to GND, VCC of the sensor to 5V and AOUT from the sensor to Signal on ANAVI Gardening uHAT.

* Are you using Raspberry Pi OS?

* Is SPI enabled? https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-gardening-uhat/anavi-gardening-uhat.md#enable-spi

* Are you running the `soil-moistore-sensors.py` example? Could you please share a few lines from the output for a reference.

Best regards,
Leon

fpy-homeb, 2022-05-13:
Hey @leon-anavi 

Thank you for quick reply

[image]

Yes spi is enabled 

pi@PlantSensor:~/anavi-examples/anavi-gardening-uhat/soil-moistore-sensors/python $ python3 soil-moistore-sensors.py
Soil Moisture Sensor 1: 100 % Soil Moisture Sensor 2: 100 %
Soil Moisture Sensor 1: 100 % Soil Moisture Sensor 2: 100 %
Soil Moisture Sensor 1: 100 % Soil Moisture Sensor 2: 100 %
Soil Moisture Sensor 1: 100 % Soil Moisture Sensor 2: 100 %
Soil Moisture Sensor 1: 100 % Soil Moisture Sensor 2: 100 %
Soil Moisture Sensor 1: 100 % Soil Moisture Sensor 2: 100 %
Soil Moisture Sensor 1: 100 % Soil Moisture Sensor 2: 100 %

leon-anavi, 2022-05-13:
OK, the wiring looks good and SPI is enabled... 
> ` pi@PlantSensor:~/anavi-examples/anavi-gardening-uhat/soil-moistore-sensors/python $ python3 soil-moistore-sensors.py Soil Moisture Sensor 1: 100 % Soil Moisture Sensor 2: 100 % Soil Moisture Sensor 1: 100 % Soil Moisture Sensor 2: 100 %`

> 
> I am controlling via ssh is that an issue?

No, login via SSH is totally fine and actually recommended.

Thanks,
Leon

fpy-homeb, 2022-05-14:
When no sensors are connected:

pi@PlantSensor:~/anavi-examples/anavi-gardening-uhat/soil-moistore-sensors/python $ python3 soil-moistore-sensors.py
Soil Moisture Sensor 1: 0 % Soil Moisture Sensor 2: 0 %
Soil Moisture Sensor 1: 0 % Soil Moisture Sensor 2: 0 %
Soil Moisture Sensor 1: 0 % Soil Moisture Sensor 2: 0 %

When 1 sensor is connected:

pi@PlantSensor:~/anavi-examples/anavi-gardening-uhat/soil-moistore-sensors/python $ python3 soil-moistore-sensors.py
Soil Moisture Sensor 1: 100 % Soil Moisture Sensor 2: 0 %
Soil Moisture Sensor 1: 100 % Soil Moisture Sensor 2: 0 %
Soil Moisture Sensor 1: 100 % Soil Moisture Sensor 2: 0 %

leon-anavi, 2022-05-18:
Hi @fpy-homeb,

So do I understand right that basically no matter which of the 2 sensors you plug it always shows 100% and if there are no sensors attached it is 0%?

These capacitive soil moisture sensors provide analog output which is converted to a digital value by MCP3008 ADC (analog-to-digital converted) after that `soil-moistore-sensors.py` maps the value to a percentage with function `valmap` https://github.com/AnaviTechnology/anavi-examples/blob/master/anavi-gardening-uhat/soil-moistore-sensors/python/soil-moistore-sensors.py#L33 The default range is between 3.5V and 5V but may be in your case due to a different environment is should be adjusted to cover larger range: https://github.com/AnaviTechnology/anavi-examples/blob/master/anavi-gardening-uhat/soil-moistore-sensors/python/soil-moistore-sensors.py#L80

Best regards,
Leon

fpy-homeb, 2022-06-01:
Hey Leon, 

Sorry life got in the way, thank you for pointing me to this. 

I have updated the script to read (sensor1, 5, 1,0,100), 0)
 And out of a fluid i am still getting a reading of 38% and in water I'm getting 87%

Soil Moisture Sensor 1: 38.0 % Soil Moisture Sensor 2: 87.0 %
Soil Moisture Sensor 1: 38.0 % Soil Moisture Sensor 2: 87.0 %

I'm guessing I can't go higher than 5 as the pi is limited to 5v? Or lower than 0V?

Thanks for you help

leon-anavi, 2022-06-01:
Hi @fpy-homeb,

> I'm guessing I can't go higher than 5 as the pi is limited to 5v? Or lower than 0V?

Yes, that's correct, the range is between 0V and 5V. Probably a tuning within that range is required in your environment to adjust the minimal value in this range that maps to 0V.

Best regards,
Leon

fpy-homeb, 2022-06-01:
Brilliant thank you! 

Ill have a play and see what i can make work

fpy-homeb, 2022-06-02:
So after a few hours of testing and some basics maths. 

My settings are 4, 1.5. 
This results in a range of 1% to 100% which i can live with.

Hope this helps other moving forward. 

I'm also looking at updating a copy of the example script to pull into InfluxDB, if you like i can post that once i have finished as well?

leon-anavi, 2022-06-02:
@fpy-homeb,

Thank you for taking the time to share the feedback. Yes, it will be great to have an example script for InfluxDB in future. 

Btw you can submit a Crowd Supply field report about your finding for the voltage range in your environment and get $25 Crowd Supply credit as a reward :) For details please have a look at: https://www.crowdsupply.com/field-reports 

Thanks,
Leon

closed 2023-01-30 by leon-anavi