AnaviTechnology/anavi-docs #48: Customer Missing Scews Needed

TannerGray1138, 2022-05-19:
Hello, 

A customer submitted a support ticket to us and stated that they were missing some pieces for their item. Their ticket is below: 

"Received ANAVI Fume Extractor (Developer Kit) yesterday. The project contains only enough screws/nuts for assembling circuit board, OLED display, and fan. Still missing screws to complete attaching light sensor and other modules. Can you ship remainder of screws and nuts?"

Is this something you can assist with? @leon-anavi 

Thank you, 
Tanner

TannerGray1138, 2022-05-19:
Customer's address: 

Jack A. Lester
5517 REYNARD TRL
STONECREST, GA 30038
United States of America
6783148511

leon-anavi, 2022-05-19:
Hi @TannerGray1138,

Please don't share private information such as shipping address as this is a public repository. I've deleted your message with private details.

ANAVI Fume Extractor contains a screw and a nut of attaching BH1750 light sensor module to the front panel. The other sensor modules should be left free on the back cover. [Here is a step by step YouTube video with assembly guidelines](https://www.youtube.com/watch?v=jscFGFjlJsU) and more [details in the user's manual](https://github.com/AnaviTechnology/anavi-docs/blob/master/anavi-fume-extractor/anavi-fume-extractor.md#assembly). The peripherals are connected to the board and the acrylic enclose on steps 8 and 9 of the video.

Best regards,
Leon

closed 2022-05-19 by TannerGray1138