AnaviTechnology/anavi-docs #50: Fume Extractor OLED display not turning on

leo-lb, 2023-06-08:
Hello!

I left my assembled fume extractor in a box with bubblewrap for a while and I just unpacked it and the OLED display is not displaying anything upon boot.

It used to work fine. All wiring is still intact and looks OK, correct pins for the display. Did it just die? How can I check that easily?

Thanks

leo-lb, 2023-06-08:
Upon further investigation it seems that when I unplug both back sensors, BMP180 and SHT21 it works again somehow.. maybe I2C address related? How could that be? How can I make these 2 back sensors work again along with it? Thanks

leo-lb, 2023-06-08:
Everything's alright now, SDA/SCL duponts were inverted on one of them. Thanks.

closed 2023-06-08 by leo-lb

leon-anavi, 2023-06-08:
Hi @leo-lb,

Thank you for feedback and for solving the issue. Have fun and fruitful work with the fume extractor. Please feel free to open a new issue if you experience any issues in future.

Thanks,
Leon