AnaviTechnology/anavi-docs #52: fauxmo issue with CommandLinePlugin object

ripface, 2024-10-20:
Hi,

I recently built a Raspberry Pi with Bookworm and installed an IR transmitter to control our lounge fire via Alexa.  It works great!!

I decided to implement a similar setup to control our electric bed so we could raise and lower them when we want to watch TV.  I had a spare Raspberry Pi 3 and loaded up Bookworm, installed lirc and fauxmo and thought this would be an easy job. 

Hmm ....

Having all sorts of problems with Fauxmo.  When I start up Fauxmo I am getting weird errors related to CommandLinePlugin that I've not seen before, and have never seen on my other pi.  

"AttributeError: 'CommandLinePlugin' object has no attribute '_latest_action'

Could not understand what was going on, so performed a clean install of bookworm, and started from scratch.  I was surprised when I completed the install that I was getting exactly the same errors reported while running fauxmo.  Have you seen anything like this before, it happens even with a small test.json with only one device in it.

Thanks!

ripface, 2024-10-20:
`2024-10-20 17:03:58 asyncio:1771     ERROR    Fatal error: protocol.data_received() call failed.
protocol: <fauxmo.protocols.Fauxmo object at 0x7f97060b10>
transport: <_SelectorSocketTransport fd=9 read=polling write=<idle, bufsize=0>>
Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/selector_events.py", line 1009, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/alexa/venv/lib/python3.11/site-packages/fauxmo/protocols.py", line 70, in data_received
    self.handle_action(msg)
  File "/alexa/venv/lib/python3.11/site-packages/fauxmo/protocols.py", line 156, in handle_action
    state = self.plugin.get_state().casefold()
            ^^^^^^^^^^^^^^^^^^^^^^^
  File "/alexa/venv/lib/python3.11/site-packages/fauxmo/plugins/commandlineplugin.py", line 166, in get_state
    return super().get_state()
           ^^^^^^^^^^^^^^^^^^^
  File "/alexa/venv/lib/python3.11/site-packages/fauxmo/plugins/__init__.py", line 98, in get_state
    return self.latest_action
           ^^^^^^^^^^^^^^^^^^
  File "/alexa/venv/lib/python3.11/site-packages/fauxmo/plugins/__init__.py", line 64, in __getattribute__
    return object.__getattribute__(self, name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/alexa/venv/lib/python3.11/site-packages/fauxmo/plugins/__init__.py", line 111, in latest_action
    return self._latest_action
           ^^^^^^^^^^^^^^^^^^^
  File "/alexa/venv/lib/python3.11/site-packages/fauxmo/plugins/__init__.py", line 64, in __getattribute__
    return object.__getattribute__(self, name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'CommandLinePlugin' object has no attribute '_latest_action'`

ripface, 2024-10-23:
Just as an update, I checked my Pi3 today and discovered that although the error messages above are still manifesting, the fauxmo package is actually working ok and I can control the beds in the bedroom using Alexa.  I guess I just need to sort out how to stop /var/log/syslog from filling up the root filesystem now ..... but it would be nice to have a resolution to this instead of managing the symptoms ....

leon-anavi, 2024-10-23:
Have you asked in fauxmo forums and GitHub repositories? The issue doesn't see directly related to the Infrared pHAT so I think fauxmo experts can give you some valuable guidelines how to fix these issues.

> I guess I just need to sort out how to stop /var/log/syslog from filling up the root filesystem now ..... 

Have you considered using `logrotate`? It rotates and compresses system logs.